bigjosh/Timed-Capsule #9: I do not like solenoids

bigjosh, 2023-12-31:
I am scared they are going to bind when the big moment comes.

Maybe go back to the original little ones so we can pull them all simultaneously? Then the interaction could be to have the user turn the locking ring again to unlock, which would lift the sliders off the solenoid shafts so they could move freely?

Or maybe just stick with current solenoids but dither them we open them one at a time but keep each one open until they are all open?

Or maybe have the user turn the ring, then hit each solenoid once with full strength to unsick it, then go back around and open them again but this time holding them open?

bigjosh, 2024-01-16:
We should at least add some teflon inserts where the solenoid shaft touches the latch edge.

bigjosh, 2024-01-16:
Test muscle wires someday.

bigjosh, 2024-02-11:
Maybe wax motors?

https://waxmotor.com/blogs/introduction-to-wax-motor

cheewee2000, 2024-02-14:
Super cool! I like how they work. But wouldn’t it be too easy to hack? Would you be able to put the whole TC into an oven and heat it up to unlock it?

bigjosh, 2024-02-14:
I think we could secure this by having counter acting actuators (CAAs?) that go in the opposite direction but are unpowered.  Super duper simplified sketch...

[image]

Let's say you heat this whole thing up. Then both A and B will actuate and cancel each other out. To get movement of the linkage, you have to, say, heat only one of the two actuators - which you would do electrically. If you heat only A then the linkage moves to the right, only B and it moves to the left. 

Make sense? I think it is secure as long as there is no way to selectively heat one or the other, which would be hard inside a thick metal shell.