bigjosh/Timed-Capsule #24: Trigger magnet is too weak

bigjosh, 2024-02-03:
Need a bigger mangent.

cheewee2000, 2024-02-06:
I put a bigger magnet in. It works, but if I spin the lever quickly, into the lock position, it doesn't trigger the switch. It seems to only trigger when I pass over the reed switch slowly.

bigjosh, 2024-02-06:
Maybe even stronger magnet? Or two weak magnets? Or a thin magnet that sits in the gap between the lever and the PCB (is there room in here?

This turns out to not be *critical* because the user can always turn the lever again until the trigger registers, but would be bad UI to have it not register under normal conditions.

cheewee2000, 2024-02-08:
It seems like the perfectly aligned magnet doesn't trigger the reed switch. So I tried moving the reed switch slightly off axis. and this seems to work reliably.

[image]

bigjosh, 2024-02-14:
That is so brutal. Hopefully in future units we can move/rotate the magnet rather than the switch? As per #14.