bigjosh/Timed-Capsule #28: Make batteries changeable from outside the locked compartment

bigjosh, 2024-02-20:
Solves so many problems. If batteries go dead during the countdown, just change them. Fail safe. 

Does require making sure that the counter is persistent across power cycles, but the support for this is already in place. 

Does require making sure that the battery compartment does not create any new physical attack vector. Maybe machine a pocket for it in the top of the lid and only tiny protected hole for wires? Or integrate the battery contacts into the sides of the pocket with an insulator to isolate them from the lid metal?

The power lines now must also have attack prevention so at least a zeener for over voltage and a self reset fuse? Maybe inductor for current fuzz resistance?

cheewee2000, 2024-03-29:
[image]
how about something like this? two pins that you an alligator clip to (or press a 9v battery against) to power it?

or what about using these battery holders? https://www.digikey.com/en/products/detail/bulgin/BX0011-1/1980754 so the battery goes in vertically from the front plate.

bigjosh, 2024-03-29:
I had imagined just normal AA battery bays, except that are accessible from the top. And they are metal and secure. I think it would be obvious to the user how/when to use these. Is there room? I think once we switch to something like muscle wire or wax motors then we can get by with only 2xAA. 

>how about something like this? two pins that you an alligator clip to (or press a 9v battery against) to power it?

That is a solution for the case where the batteries are strong enough to keep the counter going but not strong enough to open the latches, and I think it does accomplish this. But I was hoping for a more general solution where you can change the actual batteries that power both functions. Just seems cleaner to me? With this solution, I do not think it will be hard to make it clear what case those are for and when to use them to a user 100 years from now without adding several sentences of text somewhere. 

>or what about using these battery holders? 

Those are cool! Can you make them secure? Maybe put metal all around them on the inside? Is there room for all that? Maybe easier to make your own version of these out of all metal?

cheewee2000, 2024-03-29:
There's barely enough room to fit the two vertical battery holders. But I'll try! How would muscle wire or wax motors defend themselves from a heat attack?

cheewee2000, 2024-03-29:
If we make our own AA holder, could we make the top plate (and the whole scuba tank) ground? So the battery cap could just be a piece of aluminum that connects to the minus side of the battery?

bigjosh, 2024-03-30:
I think so! The only issue would be making sure that there was no way to hurt the PCB by sending signals on the externally accessible housing. I can not think of how this could be a problem, but we should ask other people if they can!

bigjosh, 2024-03-30:
How to defend against attacks on thermal actuators...
https://github.com/bigjosh/Timed-Capsule/issues/9#issuecomment-1944093638